Amp-Lab-at-VT/print-queue #4: [3d-print]: VT CRO Sweeper Arm

MinMaxolotl, 2023-03-30:
### Contact Details

kmaxwell21@vt.edu

### About Your Print

Hey there! This print should be the same dimensions as in the file and should have a 25% fill. 

### Material

PLA

Forsyth-Creations, 2023-03-30:
Thank you for submitting this ticket! Staff will look at it tomorrow and get things printed!

Forsyth-Creations, 2023-03-30:
@MinMaxolotl I don't see a part file. Did you attach it?

MinMaxolotl, 2023-03-30:
Hm, I thought there was a file attached. It should be here
[SweeperArm.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11115974/SweeperArm.zip)

Forsyth-Creations, 2023-03-30:
Sweet, I'll get this started sometime tomorrow!

Forsyth-Creations, 2023-03-31:
@MinMaxolotl this is still just the Solidworks file. The Amp Lab does not make a habit of going in and grabbing your mesh files from the part. Please submit them in either an STL or 3MF format. Thank you!

MinMaxolotl, 2023-03-31:
Ahh once again I apologize. I am unfamiliar with making prints, here is HOPEFULLY the correct version. 
[SweeperArm.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11117684/SweeperArm.zip)

Forsyth-Creations, 2023-03-31:
Printing in-lab right now

Forsyth-Creations, 2023-04-03:
I don't see this in-lab. Was it picked up? If so, @MinMaxolotl please mark this issue as closed

MinMaxolotl, 2023-04-03:
Sorry! Just saw this. Yeah I've been coming in and picking up the prints when I get the notifications they are complete. Thank you!

closed 2023-04-03 by MinMaxolotl

Forsyth-Creations, 2023-04-03:
> Sorry! Just saw this. Yeah I've been coming in and picking up the prints when I get the notifications they are complete. Thank you!

@MinMaxolotl 

No problem! Just to keep the loop closed, if you are picking up the prints, make sure to update the status of the issue! Thank you for helping us test this new 3D printing process lol