Amp-Lab-at-VT/print-queue #6: [3d-print]: Servo Sweeper Arm and Servo Head

MinMaxolotl, 2023-04-03:
### Contact Details

kmaxwell21@vt.edu

### About Your Print

The servo arm is 192 mm long, 87 mm tall, and 5 mm thick. 25% fill. We need just one.
The servo head (Sweeper Support) is 45 mm long, 25 mm, and 5 mm thick. We need three of them at 30% fill. 
Thank you so much for the support! 
[SweeperSupport.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11142507/SweeperSupport.zip)

### Material

PLA

Forsyth-Creations, 2023-04-11:
Hey there! Just wanted to give an update on this. I haven't had a chance to swing into the lab to tend to the Bambu, as it ran into a print issue last week. I've now had a chance to inspect the machine, and have started the print! Assuming all goes well, it should be in-lab ready for pickup around 1:00. I'll keep you posted

Forsyth-Creations, 2023-04-11:
ETA seems to be 12:15

Forsyth-Creations, 2023-04-11:
Ah, I just noticed you need three arms. One is currently printing, and I will run the last two once this print is complete

Forsyth-Creations, 2023-04-11:
If you make your way to the lab, all your parts should be finishing up. Please close this issue when you have picked up your parts!

MinMaxolotl, 2023-04-12:
Got it! Thank you

MinMaxolotl, 2023-04-12:
I still need to pick up the extras, but I can grab them tomorrow

closed 2023-04-12 by MinMaxolotl