Amp-Lab-at-VT/print-queue #7: [3d-print]: VT Cookie Cutters

richgibbons98, 2023-04-29:
### Contact Details

gricha1@vt.edu

### About Your Print

These are some test prints for VT Cookie Cutters
[El-Ultimo-Cookie-cutter-Apr-29.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11360161/El-Ultimo-Cookie-cutter-Apr-29.zip)

### Material

PLA

Forsyth-Creations, 2023-05-01:
Starting this print now

Forsyth-Creations, 2023-05-05:
@richgibbons98 I know this print finished, but it's still protocol for me to check in. How'd it turn out? If you like it, feel free to close this issue ticket :)

richgibbons98, 2023-05-06:
It turned out great! Just waiting on the cookies at this point.

Best,
Richard Francis Gibbons III
Graduate Teaching Assistant - MDE
Virginia Tech Electrical Engineering Class of 2021
571-455-9877
***@***.***

richgibbons98, 2023-05-06:
How do I close the issue ticket?

Best,
Richard Francis Gibbons III
Graduate Teaching Assistant - MDE
Virginia Tech Electrical Engineering Class of 2021
571-455-9877
***@***.***

Forsyth-Creations, 2023-05-06:
When you leave a comment, you should be able to click the "Close with Comment" button, which will close the issue. I can do it, but I want you to get the exposure to it.

Forsyth-Creations, 2023-05-06:
[image]

richgibbons98, 2023-05-06:
done

closed 2023-05-06 by richgibbons98