Amp-Lab-at-VT/print-queue #8: [3d-print]: My very cool building idea (it has to do with electronics)

epritchard13, 2023-05-01:
### Contact Details

epritchard@vt.edu

### About Your Print

This is a building. It would be cool to 3d print a scale model. There will be lighting installed into the building in order to take photos of the lighting and experimenting in wiring.
[cool building.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11363269/cool.building.zip)

### Material

PLA

Forsyth-Creations, 2023-05-01:
I don't believe the comment, but I'll print it anyway

Forsyth-Creations, 2023-05-01:
@epritchard13 Did you get this print off the bed? I would like to start @richgibbons98 print

Forsyth-Creations, 2023-05-01:
Richard has confirmed this print has been completed, or at the very least that it was removed from the bed. Closing this issue

closed 2023-05-01 by Forsyth-Creations