Amp-Lab-at-VT/print-queue #9: [3d-print]: void keyboard switch test

sakgoyal, 2023-05-27:
### Contact Details

sakgoy2001@vt.edu

### About Your Print

I have some stl files I downloaded that I want to print. No idea what material it should be but I think any should be fine. 
since this part is small and has small details, I would prefer a high amount of precision/accuracy

total dimensions approximate: (~3.3cm x ~3cm x ~1cm)

I would like to get 4 of each of the 2 stl files please. [void_switch_test_print.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11581859/void_switch_test_print.zip)

[image]
[image]

### Material

Don't Know

Forsyth-Creations, 2023-06-19:
We've been slow to respond due to the summer, but did this get printed? Wanted to make sure so I can get it printed for you

Forsyth-Creations, 2023-08-01:
@sakgoyal Does this still need to be printed?

sakgoyal, 2023-08-02:
Yes please

Forsyth-Creations, 2023-08-02:
This has been printed, and is now awaiting you in-lab. Let me know when you pick it up, and I will close out this issue!

Forsyth-Creations, 2023-10-17:
Was this picked up @sakgoyal? Just want to confirm if I can close this

sakgoyal, 2023-10-20:
Yep

closed 2023-10-20 by sakgoyal