Amp-Lab-at-VT/print-queue #10: [3d-print]:

RaymondWei78, 2023-06-19:
### Contact Details

wraymond03@vt.edu

### About Your Print

Fiber optic adaptor for testing convex lens focusing beams. 
[FiberTestingAdaptor.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/11791350/FiberTestingAdaptor.zip)

### Material

PLA

Forsyth-Creations, 2023-06-19:
I'll see if I can start this tonight!

Forsyth-Creations, 2023-06-20:
@RaymondWei78 I'm getting this printed at my place and will drop it off in-lab. I couldn't remember if I had re-loaded the filament in-lab on the P1P, so this seemed like the quickest solution. 

Two quick things:
1. I'd like to train you to start prints at some point, that way you can have unimpeded access. 
2. I don't want us to make a habit out of printing things for other labs. However, you've put a lot of work in for us with the CNC, so I'd like to return the favor. 

Print has been started, and I will try to deliver later this morning!

RaymondWei78, 2023-06-20:
I’d really appreciate that! Please let me know how you plan on making that happen, thanks Henry!

Forsyth-Creations, 2023-07-10:
I believe I finished this

closed 2023-07-10 by Forsyth-Creations