Amp-Lab-at-VT/print-queue #12: [3d-print]:

devincress, 2023-10-03:
### Contact Details

dcress94@vt.edu

### About Your Print

SMD component magazine w/ rail for the lab. 65x67x15mm and 120x70x20mm.

[Compliant_SMD_Magazine_15_2.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/12796950/Compliant_SMD_Magazine_15_2.zip)

[Rail_120-5.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/12796987/Rail_120-5.zip)

### Material

PLA

Forsyth-Creations, 2023-10-17:
how many?

Forsyth-Creations, 2023-10-17:
For viz, I've started printing this in the lab. I figure we can print more if we need them, and the files will be available. The ETA is about 5 hours

Forsyth-Creations, 2023-10-20:
@devincress if you're happy with what was printed, please feel free to close this issue

Forsyth-Creations, 2023-10-22:
I know these have been printed, so I am going to close this issue. @devincress please close these issues in the future to provide better traceability

closed 2023-10-22 by Forsyth-Creations