Amp-Lab-at-VT/print-queue #16: [3d-print]: Prints for Stylophone

jiviteshkuk26, 2023-11-24:
### Contact Details

jiviteshkuk26@vt.edu

### About Your Print

Description
->Lower Body, (202x80x13.60) mm, Black
->Upper Body, (202x80x12.50) mm, Black
->Filler Plate, (199x35.50x1.9) mm, Red
->Face Plate, (202x38.50x3.9) mm, Black

Files:
[3D Print Files.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/13461492/3D.Print.Files.zip)

Thank You!

### Material

PLA

Forsyth-Creations, 2023-12-12:
@jiviteshkuk26 I did print this. If this is acceptable, please close this issue :) I will close in a week if you don't respond

jiviteshkuk26, 2023-12-13:
> @jiviteshkuk26 I did print this. If this is acceptable, please close this issue :) I will close in a week if you don't respond

I kept it open since I wanted to print everything again as most of the current body broke :(. I will close it for now and print it again after the break. Thank You, Have Fun!

closed 2023-12-13 by jiviteshkuk26