Amp-Lab-at-VT/print-queue #25: [3d-print]: camera bracket for cubesat

jtjagoda, 2024-03-22:
### Contact Details

jtjagoda@vt.edu

### About Your Print

Dimensions: 4.39cm x 8.8cm x 1cm
[camera bracket.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/14726762/camera.bracket.zip)

### Material

PLA

Forsyth-Creations, 2024-04-02:
This print is complete and is waiting for you in-lab. Let me know if there are any issues, and please close this ticket after pickup!

jtjagoda, 2024-04-02:
Hello, I am having trouble finding the print. Is there some location I should be looking for it? It appears there is another print going on right now.

Thanks,

JT Jagoda
inspireFly Project Manager | IC CAE Scholar
Virginia Tech | Mechanical Engineering
(860)-798-1940 | ***@***.***

***@***.***?anonymous&ep=bwmEmailSignature>           Book time to meet with ***@***.***?anonymous&ep=bwmEmailSignature>

closed 2024-04-02 by jtjagoda