Amp-Lab-at-VT/print-queue #33: S24-12 Senior Design Robot printing

sarahfox03, 2024-04-01:
### Contact Details

sarahfox@vt.edu

### About Your Print

These are parts that we need printed for our senior design project! Please let me know if i should have uploaded them separately. Thank you SO much!

[3D printed parts Senior Design Swarm Robotics.zip](https://github.com/Amp-Lab-at-VT/print-queue/files/14828136/3D.printed.parts.Senior.Design.Swarm.Robotics.zip)
project.

### Material

PLA

Forsyth-Creations, 2024-04-02:
This print is complete and is waiting for you in-lab. Let me know if there are any issues, and please close this ticket after pickup!

Forsyth-Creations, 2024-04-14:
This is printed and awaiting for you in-lab. Please close this issue after pickup

Forsyth-Creations, 2024-04-26:
Closing this issue due to inactivity

closed 2024-04-26 by Forsyth-Creations