Amp-Lab-at-VT/print-queue #45: [3d-print]: Senior Design F12 Swarm Robotics Bot

sarahfox03, 2024-10-10:
### Contact Details

sarahfox@vt.ed

### About Your Print

[3d print 1010.zip](https://github.com/user-attachments/files/17335253/3d.print.1010.zip)

Sorry for so many parts but I thank you in advance! If its easier for me to condense these i definitely can. Thanks!!

Here are the quantities: 
chassis                         x1
connector                    x2
holding board 9v         x1
ir sensor a                   x1
ir sensor b                  x1
master wheel             x2

mechanical track       x32
slave wheel                x2
slave wheel SL           x2
ultrasonic 1               x3
ultrasonic 2a             x3
ulrasonic  2b             x3

### Material

PLA

Forsyth-Creations, 2024-10-17:
This print has been started in-lab, and will be complete in 10 hours. I will check on it and see if there are any issues that might change this timeline. Sorry for the delay! Didn't see this request until today

Forsyth-Creations, 2024-10-17:
[Car.zip](https://github.com/user-attachments/files/17416357/Car.zip)
Here's the slice file I used btw in case we need to re-print. It was generated with Bambu Labs software (free if you want to take a shot at downloading it and seeing for youself)

closed 2024-11-15 by Forsyth-Creations