Amp-Lab-at-VT/print-queue #49: [3d-print]: add on to issue 45, Car

sarahfox03, 2024-11-05:
### Contact Details

sarahfox@vt.edu

### About Your Print

Hello!
I need five of these (sorry ) but i need a different size chassis. I have attached. If I could have these as quickly as possible, that would be awesome , for senior design. Thank you SOO much
[Car.zip](https://github.com/user-attachments/files/17633184/Car.zip)
[](url)
[smars_chassis_SL_tall_3mm.zip](https://github.com/user-attachments/files/17633193/smars_chassis_SL_tall_3mm.zip)

### Material

PLA

Forsyth-Creations, 2024-11-15:
Starting this on my home printer now. Should be able to churn out all five by tomorrow afternoon. Sorry for the delay!

Forsyth-Creations, 2024-11-18:
[image]

Forsyth-Creations, 2024-11-18:
They're in-lab in the boxes!

closed 2024-11-18 by Forsyth-Creations